ButterFlight/butterflight-configurator #16: Tramp Telemetry/Smart Audio

PeteG-FPV, 2018-03-17:
Revolt FC running ButterFlight 3.4.2 with Tramp Telem on Uart 6. Seems that the FC was always putting the Tramp into IRC Channel 1 25mW. Using the push button on Tramp or TnR wand to change channels worked for about 1secs then reverted back to IRC Channel1.

Tried multiple tramps with and without TNR boards, worked as soon as I cut the telemetry wire going to UART6.

Flying-Potatoes, 2018-03-17:
JB F4, Butterflight 3.4.2, Unify Pro HV on soft serial. Same issue. But setting VTx via Butterflight osd rather than Taranis/smart audio works fine.

Clynikal, 2018-03-30:
SA working on LUA here on 3.5.0 & 10.3.0 BuF

beezweks, 2018-03-31:
hey guys im on butter 3.5 config 10.3.0 butter lua script loaded on my taranis x9d 2.2.0.. lua script is on the sd card the bfl and the script file but its not found when i want to open it on the script page it says... no scripts on sd card.... what im i doing wrong here

LexioTech, 2018-04-05:
@PeteG-FPV Hi, are you sure you are using the (TX) pad of Uart 6 for the feature? If it's correct, then please try using soft serial on a free pad and use that for IRC Tramp

@Flying-Potatoes Hi, try upgrading to BuF 3.5.0 and try it with the Legacy Smart Audio feature on / off. People have varying issues with different TBS unifys that work either with or without this feature.

@beezweks Hi, Please try to follow this guide for installing the Lua scripts: https://github.com/ButterFlight/butterflight-tx-lua-scripts/blob/master/README.md

Also, make sure that your openTX 2.2.0 version has lua enabled. You can check that in the Open TX companion app when flashing/creating a firmware

Flying-Potatoes, 2018-04-05:
@LexioTech  Thanks, will try. Not a problem given that osd control works.

LexioTech, 2018-04-05:
@Flying-Potatoes Ohh sorry misunderstood your issue a bit at 3AM, so if you have issues with the LUA, try checking the solution I suggested for @beezweks :)

beezweks, 2018-04-05:
Thanks.. It's fixed.

kidBrazil, 2018-04-10:
Closing as fixed

closed 2018-04-10 by kidBrazil