ButterFlight/butterflight-configurator #21: Can't enable DSHOT Beacon Configuration slider with Beacon Tone #1 selected.

richard-scott, 2018-04-02:
I'm using release https://github.com/ButterFlight/butterflight-configurator/commit/76045762ad44cf116faf120f43ff9506b393f7e0 from Github of the BuFC and I have an AnyFC F7 board and once I've disabled DSHOT Beacon Configuration in the Configuration tab I am unable to re-enable it easily.

[image]

When I try, I get slightly different results to the [BetaFlight Configurator](https://github.com/betaflight/betaflight-configurator/issues/975) as the FC Reboots, but once I go back in and find that the DSHOT Beacon is still not enabled.

The only way I can enable it is to set the "Beacon Tone" to '2'. Then it works with a reboot each time. I can set tones 2 - 5 ok, and once setting the tone [2-5] I can then go back and set it to 1.

kidBrazil, 2018-04-12:
@richard-scott thanks for the heads up richard - definitely seems a bit odd and we will look into it.

SpeedMetal1, 2018-05-19:
Same issue here  When Dshot Beacon Configurator is selected and Beacon Tone set to 1 it will not save.  But set to any other tone it works.