ButterFlight/butterflight-configurator #27: can't install 3.5 on windows 10 x64

S-T-A, 2018-04-05:
I'm getting "Butterflight Configurator Setup Warning"
C:Program
Files\Butterflight\Butterflight-Configurator\uninbf00.dat is missing, some elements could not be removed. These can be removed manually.

LexioTech, 2018-04-06:
@S-T-A Hi, I assume you had a previous version of the configurator installed, 

- Did you remove that manually? By deleting its folder or something?
- If you had a version installed, what version? 32 or 64 bit? 
- If your installation still exists, could you please create a screenshot of the folder ( C:\Program Files\Butterflight\Butterflight-Configurator\) contents with all files visible?
- As a workaround, you can try and manually delete the configurator's folder (C:\Program Files\Butterflight\Butterflight-Configurator\) and then try the setup again.

kidBrazil, 2018-04-12:
@S-T-A So install i sworking fine but you have some spacing issues with OSD? Can you send us a screenshot?

S-T-A, 2018-04-13:
Will do this week end

closed 2018-04-17 by kidBrazil