ButterFlight/butterflight-configurator #30: Can't install on Ubuntu 16.04LTS

Padraignotlad, 2018-04-12:
When I try to install 10.4.0 I get an error during installation.  It fails.  So I thought I'd try 10.3.0 and that doesn't give an error, but it won't install either.  Just kind of does nothing.  If you click Install a 2nd time, it will flicker and change to give you the option to remove the software.  But I don't think it actually installs.  I can't find it.  - Padraig

kidBrazil, 2018-04-12:
@Padraignotlad - For now I would suggest using the Linux 64 Zip and just extracting it somewhere. You can then run it via terminal or create a shortcut for it.

We will be looking into what is broken with deb

Padraignotlad, 2018-04-12:
Already up and running.  Thanks!

kidBrazil, 2018-04-13:
This issue apparently was inherited from Betaflight - we are just going to stop supplying Deb packages for now.

closed 2018-04-13 by kidBrazil

SidSilver, 2018-05-02:
The issue with ButterFlight on linux is just the name of the file butterflight-configurator.desktop
It is misnamed "betaflight-configurator.desktop" instead of "butterflight-configurator.desktop"
Just rename it and it works.