ButterFlight/butterflight-configurator #35: Avast flags  installer as malware.

Freakofspeed, 2018-04-17:
butterflight-configurator-installer_10.4.1_win64.exe
Was flagged as FileRepMalware by Avast Business Antivirus 18.2.2530
Probably a false positive.

kmitchel, 2018-04-17:
Quick Google search, and it looks like Avast has a long history of flagging
NSIS installers.

http://nsis.sourceforge.net/NSIS_False_Positives

kmitchel, 2018-04-17:
I've reported the false positive to Avast.  We'll see if they fix it.

kmitchel, 2018-04-17:
Avast emailed back. Please update your virus definitions, and try again.  I
wouldn't be surprised if it takes a day or two.

Freakofspeed, 2018-04-26:
butterflight-configurator-installer_10.4.2_win64.exe
Passed Avast Business Antivirus 18.2.2530 Virus Definions 180425-6

Cheers!

closed 2018-04-26 by Freakofspeed