ButterFlight/butterflight-configurator #36: Stage 2 and lowpass gyro filter options not showing

GerardMatthews, 2018-04-17:
The filter options do not show if you already have expert mode enabled. If you then disable and re-enable them the stage 2 filter shows but still not the low past filter option. Please see screenshots showing problem.

I have BF 3.5 installed on the quad with fast kalman enabled already.

[image]

After disabling and then re-enabling I can see the stage 2 filter options.

[image]

nefer84, 2018-04-26:
On configurator 10.4.2 i cant see stage2 filter . In was able to ser on 10.4.1 but always show nome. Even after Change it to kalman (default) and save it, after reboot ThE configurator always show none.

LexioTech, 2018-04-26:
@nefer84 Are you running a Helio board?

nefer84, 2018-04-26:
@LexioTech yes um.using helio board

LexioTech, 2018-04-26:
@nefer84 Those filtering options were removed from the configurator in case of Helio boards. You can edit the built in imuf settings on the filters tab though. You might need to upgrade to the latest rc version fw for that.

nefer84, 2018-04-26:
@LexioTech thank you.
What about other boards, stage2 on filter tab always show none.