ButterFlight/butterflight-configurator #37: FEATURE REQUEST - Vbat THROTTLE compensation

quiljw01, 2018-04-19:
Sorry i don't github good.... Not sure how to do a feature request......

This feature request was turned down by Betaflight devs about a year ago.  Hoping Butterflight will have interest.  This feature would allow pilots to be more consistent in their flying because it would aim to keep a constant thrust to throttle stick position ratio throughout an entire battery.

I would call it Vbat THROTTLE compensation.  Basically changes a throttle curve live and according to the battery voltage.  It's essentially trying to always keep full throttle at a certain POWER level... aka, thrust level.  It would adjust this throttle curve live, maybe each PID loop, or could probably be fine if 100hz.

The pseudocode would be something like, at full vbat, limit the max throttle value to be AA%  , and at BB voltage, allow max throtle.  Where AA and BB are variables which would be something like... AA = 80% and BB would be something like 14V for 4s and 20V for 6s.  It would basically be a live adjusting linear throttle curve.  

People will say "isn't it slowing me down at full throttle?"   , the answer is technically YES..   but it will make you more consistent throughout your flight, which is proven to give you a faster race time.   So, to hell with these people!  :D   I want the feature because i think it will make me more consistent, and therefore faster.  

Thoughts? Questions? Suggestions? 

JROD

GerardMatthews, 2018-04-19:
Surely if you just limit your max throttle you will achieve much the same thing. That sounds like a really complex solution to a simple problem. E.G. You could limit your throttle to 75 or 80 percent of your throttle. It will keep you out of the top section of your throttle which can be the most draining.

The main problem I see with this though is where you actually might need that very last bit to pip the other guy. I don't race but could you not train yourself to just use less throttle as well. Otherwise you could just do my original idea.

I don't really see many developers jumping at adding this feature.

quiljw01, 2018-04-19:
Gerard, actually a simple throttle limit will surely not accomplish this
because as you near the end of the battery you actually will still get less
power than you had at the beginning of the battery. Also in times of heavy
battery sag it will still change with a simple limit like you suggest.  The
goal for racing is to be consistent.  This feature is geared towards
racers.

oyvinla, 2018-04-24:
It is an interesting feature. I'm not a great racer, but I really feel the inconsistency in the throttle sensitivity when changing packs and going from an almost depleted pack to a fresh pack. It would be interesting to try a feature like that

sparrow082, 2018-05-11:
You're not supposed to but you can already do that with the taranis. There are several problems doing it this way but the most extreme is that if you add a curve in your throttle Channel the channel no longer goes to - 100 and adding offset will not fix the problem.