ButterFlight/butterflight-configurator #57: GNU/Linux Installer has desktop application file called betaflight-configurator.desktop

blackmennewstyle, 2018-06-26:
Hi guys :)

I was trying to deploy the GNU/Linux installer on my GNU/Linux Debian and it appears that you forgot to rename the **betaflight-configurator.desktop** file to **butterflight-configurator.desktop**

meoso, 2019-05-04:
@blackmennewstyle (ABMNS), i found the same issue. i did this:
```
mv /opt/butterflight/butterflight-configurator/betaflight-configurator.desktop /opt/butterflight/butterflight-configurator/butterflight-configurator.desktop
xdg-desktop-menu install /opt/butterflight/butterflight-configurator/butterflight-configurator.desktop
```
but would prefer the packaging fixed.

blackmennewstyle, 2019-05-05:
> @blackmennewstyle (ABMNS), i found the same issue. i did this:
>
> ```
> mv /opt/butterflight/butterflight-configurator/betaflight-configurator.desktop [...]

Nice job, i managed to fixed it on mine as well but like you said a package fixed would have been better. But i'm afraid Butterflight is partially dead currently.

Have a great weekend and happy flying!